MangoAutomation/ma-core-public #989: Point Templates don't seem to import on new page

Joelhaggar, 2017-04-14:
I'm using this json and they import fine and fast on the legacy page but I only get feedback on the first one template on the new page and then it just sits there:

{
   "templates":[
      {
         "xid":"Alarm_Inputs",
         "name":"Alarm_Inputs",
         "templateType":"DATA_POINT_PROPERTIES",
         "dataType":"BINARY",
         "rollup":"NONE",
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"INSTANT",
         "purgeType":"YEARS",
         "plotType":"STEP",
         "chartColour":"blue",
         "chartRenderer":{
            "type":"TABLE",
            "limit":10
         },
         "defaultCacheSize":1,
         "defaultTemplate":false,
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":15,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "preventSetExtremeValues":false,
         "purgeOverride":false,
         "purgePeriod":1,
         "setExtremeHighLimit":1.7976931348623157E308,
         "setExtremeLowLimit":-1.7976931348623157E308,
         "textRenderer":{
            "type":"BINARY",
            "oneColour":"#ff0000",
            "oneLabel":"Alarm Active",
            "zeroColour":"#808080",
            "zeroLabel":"None"
         },
         "tolerance":0.0,
         "readPermission":null,
         "setPermission":null
      },
      {
         "xid":"Alphanumeric_Default",
         "name":"Alphanumeric",
         "templateType":"DATA_POINT_PROPERTIES",
         "dataType":"ALPHANUMERIC",
         "rollup":"NONE",
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"INSTANT",
         "purgeType":"YEARS",
         "plotType":"STEP",
         "chartColour":"red",
         "chartRenderer":{
            "type":"TABLE",
            "limit":10
         },
         "defaultCacheSize":1,
         "defaultTemplate":true,
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":15,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "preventSetExtremeValues":false,
         "purgeOverride":false,
[... 1219 lines cut ...]
         "defaultTemplate":false,
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":1,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "preventSetExtremeValues":false,
         "purgeOverride":false,
         "purgePeriod":2,
         "setExtremeHighLimit":1.7976931348623157E308,
         "setExtremeLowLimit":-1.7976931348623157E308,
         "textRenderer":{
            "type":"ANALOG",
            "useUnitAsSuffix":true,
            "unit":"",
            "renderedUnit":"",
            "format":"0"
         },
         "tolerance":0.0,
         "readPermission":null,
         "setPermission":null
      }
   ]
}

jazdw, 2017-04-17:
Please test again I believe this was fixed.

Puckfist, 2017-04-19:
I tested, did not observe issue.

closed 2017-04-21 by Puckfist