MangoAutomation/ma-core-public #991: NPE when using JsonDataDao in module install() method

jazdw, 2017-04-20:
```
ERROR 2017-04-20T10:42:13,042 (com.serotonin.m2m2.Main.main:125) - Error during initialization 
java.lang.NullPointerException: null
	at com.serotonin.m2m2.rt.event.type.AuditEventType.raiseEvent(AuditEventType.java:162) ~[classes/:?]
	at com.serotonin.m2m2.rt.event.type.AuditEventType.raiseChangedEvent(AuditEventType.java:127) ~[classes/:?]
	at com.serotonin.m2m2.db.dao.AbstractDao.update(AbstractDao.java:339) ~[classes/:?]
	at com.serotonin.m2m2.db.dao.AbstractDao.save(AbstractDao.java:302) ~[classes/:?]
	at com.serotonin.m2m2.db.dao.AbstractDao.save(AbstractDao.java:289) ~[classes/:?]
	at com.infiniteautomation.ui.UIInstallUpgrade.installMenuData(UIInstallUpgrade.java:56) ~[ui-3.0.1-SNAPSHOT.jar:?]
	at com.infiniteautomation.ui.UIInstallUpgrade.install(UIInstallUpgrade.java:33) ~[ui-3.0.1-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.module.Module.install(Module.java:136) ~[classes/:?]
	at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:271) ~[classes/:?]
	at com.serotonin.m2m2.Main.main(Main.java:121) [classes/:?]
INFO  2017-04-20T10:42:13,046 (com.serotonin.m2m2.Lifecycle.terminate:332) - Mango Lifecycle terminating... 
```

jazdw, 2017-04-20:
Due to Lifecycle calling `runtimeManagerInitialize()` after `module.install()`

Puckfist, 2017-04-20:
I don't see any problem with reordering the utilities and events initialize calls to before the postDatabase call. I reviewed modules' postDatabase, install and upgrade methods, as well as utilitiesInitialize and eventManagerInitialize. We'll see what Terry thinks...

terrypacker, 2017-04-20:
I would suggest using a startup task for this as is done in the Internal Data Source InternalMenuItem.install() this will run after Mango is fully initialized.

terrypacker, 2017-04-20:
Arguments could be made for moving around the methods in the Lifecycle but lets not consider doing that until after the release.

jazdw, 2017-04-21:
I would suggest that we

- add more lifecycle stage hooks
- add arguments to the methods so they can tell if this is the first run or upgrade at any stage

jazdw, 2017-05-23:
Closed via a711439b001855d447882fad1030ec1ab2108864

closed 2017-05-23 by jazdw

jazdw, 2017-05-30:
Tested, works fine.