MangoAutomation/ma-core-public #995: Inconsistencies between "System event alarm levels" and values in DB

jazdw, 2017-05-01:
While creating the new UI pages for system settings I noticed that the "System event alarm levels" section on the old UI was displaying values which were not consistent with what was stored in the database.

In particular the "System startup" setting was displaying "None" but the database key "systemEventAlarmLevel.SYSTEM_STARTUP" had a value of "1" which should be information. My new page displayed the correct value of "Information" which was retrieved over REST.

We also noticed that after changing some of the settings then deleting the rows from the DB the UI still displayed the previously set values instead of going back to the defaults.

jazdw, 2017-05-02:
Seems to be due to the REST API using a different SystemSettingsDao instance (and hence cache). We should use a singleton instance of the Dao and make sure the cache is thread safe.

terrypacker, 2017-05-17:
The dao is now a singleton but synchronization has not been taken into account yet.

terrypacker, 2017-05-17:
The cache is now synchronized via ConcurrentHashMap

jazdw, 2017-05-30:
Tested, seems to work fine now.

commit fe838d62 (2017-05-30): Use atomic operations on system settings cache map

Closes #995
closed 2017-05-30 by jazdw