MangoAutomation/ma-core-public #996: Neither startup or upgrade hooks are called if first start fails

jazdw, 2017-05-02:
I just started up Mango from a fresh bundle zip and noticed that the internal DS didn't create a data source. 

@Puckfist and I determined that this happened because neither the startup or upgrade hook was called.

The issue seems to have occurred due to Mango failing to fully start the first time. I had another Mango running on the same port on the first start, so it unpacked the module and then exited. On the second start Mango didn't know that it was doing a startup or an upgrade because the module was already unpacked.

This could cause issues for any other modules which use the startup/upgrade hooks, not just the internal data source.

Puckfist, 2017-10-03:
This will be fixed by #1027

jazdw, 2017-10-04:
It will? How?

Puckfist, 2017-10-05:
At the same time as*