MangoAutomation/ma-core-public #997: REST end points needed for system settings page

jazdw, 2017-05-02:
There are a lot of things on the legacy UI system settings page which are not really system settings per se. This means that we can't read or write to them using the existing /rest/v1/system-settings/ end point.

Some should probably be moved into a different page altogether e.g. Database size, Filedata size, Total size etc. should live on a system status / information page.

Here is a list of things on the system settings page I currently can't read/write to:
- Time zone
- Database type
- Database size (including "Refresh" button & "Purge now using purge settings" button)
- Filedata size
- Total size
- History count
- Top points
- Event count
- All of the "Virtual serial ports" section
- All of the Purge buttons
- The excel reports purge buttons
- The reports purge buttons
- Mango REST Api Settings CORS headers (can just ignore this one)
- More to come...

I'm not going to touch the "Permissions" section just yet either as I believe this may be subject to change.

terrypacker, 2017-05-10:
There is a lot of logic in the SystemSettingsDwr that is not part of the REST controller.  Sometimes there are additional actions that need to be performed.  This will need to be addressed via SystemSettingsListeners.  We will also need to review all the SystemSettingsModuleDefinition(s)

See #1000

terrypacker, 2017-05-12:
We should add the the following as ReadOnlySetting(s)

Reports NoSQL database size
Excel Reports File Data Size and Count

terrypacker, 2017-05-13:
The Serial Ports Section isn't even in the system settings at all, its a whole different Dao etc.  I think we will need a separate endpoint for that stuff.

terrypacker, 2017-05-15:
List of All Read Only Settings
- [ ] Reports NoSQL database size
- [ ] Excel Reports File Size
- [ ] Excel Reports File Count
- [x] SQL Database size
- [x] SQL Database Type
- [x] Filedata Count
- [x] Filedata Size
- [x] Timezone
- [x] History Count
- [x] Top Points
- [x] Events Count

List of All Actions that need to be implemented:

- [ ] Purge all point values
- [ ] Purge all events
- [ ] Purge using purge settings
- [ ] Reload Log4J Configuration
- [ ] Test Log4J debug, info, warn, error, fatal messages
- [ ] Backup Configuration Now
- [ ] Backup SQL Now
- [ ] Restore SQL Now
- [ ] Get SQL Backup List
- [ ] System Event Types Updated
- [ ] Audit Event Types Updated

List of Extra Features/Endpoints
- [ ] JSON File Upload
- [ ] JSON File Check Path Exists
- [ ] MangoES Reboot
- [ ] MangoES Set Hostname
- [ ] MangoES Set Timezone
- [ ] MangoES Set Interface
- [ ] MangoES Shutdown? (DWR Stop method)
- [ ] CRUD for Virtua Serial Ports

terrypacker, 2018-01-11:
Because of the outstanding items that don't fit nicely into the SystemAction API we may want to expand the API to not have to run a WorkItem to process the action.

terrypacker, 2019-01-04:
Closing as all have been done.

closed 2019-01-04 by terrypacker