MangoAutomation/ma-core-public #1000: SystemSettingsRestController - may not match DWR's responsibilities

Puckfist, 2017-05-08:
I observed in writing a response about the thread pools that these are explicitly set in the DWR to the timer's executor service, and not through the system settings listener. This is just to note that, as there may be other behaviors like that.

terrypacker, 2017-12-19:
We have moved all the functionality into SystemSettingsListeners.  There is special handling for thread pool settings so they don't get fired in a separate thread.

terrypacker, 2017-12-22:
It seems that all the actions are covered with SystemSettingsListeners.  However there may be some missing validation.  The validation for the RestController is done via SystemSettingsDao.validate and the DWR uses validation within its various save methods.

Things that need validation in the SystemSettingsDao are:

- [ ] LANGUAGE
- [ ] Should we remove any entries that are set to null so they don't get saved into the database?
- [ ] Http Client settings
- [ ] Email Settings
- [ ] Future date limit period type
- [ ] httpds prologue (Should be a system setting definition?)
- [ ] httpds epilogue (Should be a system setting definition?)
- [ ] instance description
- [ ] PUBLICLY_RESOLVABLE_BASE_URL
- [ ] We should not allow settings some settings i.e. NEW_INSTANCE, DATABASE_SCHEMA_VERSION
- [ ] PERMISSION_DATASOURCE
- [ ] SITE_ANALYTICS_HEAD
- [ ] SITE_ANALYTICS_BODY
- [ ] UPGRADE_VERSION_STATE
- [ ] HIERARCHY_PATH_SEPARATOR
- [ ]

terrypacker, 2018-04-10:
This is up for discussion still: Should we remove any entries that are set to null so they don't get saved into the database?

Puckfist, 2018-04-17:
I'm going to add validate for alarm levels....

Puckfist, 2018-04-17:
In regards to "Should we remove any entries that are set to null so they don't get saved into the database?" I think generally it should be okay to save a null into the database. The default value will just be used afterward, I would think.

Puckfist, 2018-04-17:
Not sure what's to be validated for PERMISSION_DATASOURCE

Puckfist, 2018-04-17:
Similar state for the site analytics. On the one hand, it would be nice to validate those since you can bork all the old pages in a single go, but on the other hand trying to validate that it's not malicious or malformed HTML or JavaScript is probably not necessary.

closed 2018-05-01 by Puckfist