MangoAutomation/ma-core-public #1002: Mark Module as Signed or Unsigned

terrypacker, 2017-05-11:
When loading modules we know if they have been signed i.e. retrieved from the store, we should add a boolean into the Module object so that we can display this information in the UI.

Puckfist, 2017-06-27:
I do not believe the Core bears any signature even if gotten from the store

terrypacker, 2017-06-27:
It is hard coded in the ModuleRegistry getCoreModule() which we should change to check one of the signed files within the core.  I think there is another issue around this exact thing.

terrypacker, 2017-10-12:
Put logic into ModuleRegistry around line 155 to check if the file release.properties is signed i.e. release.signed.

terrypacker, 2017-12-22:
Confirmed this is working in 3.3.0, downloaded from test store and saw the signed indication on the modules page.

closed 2017-12-22 by terrypacker