MangoAutomation/ma-core-public #1007: Test REST CRUD for Event Detectors

terrypacker, 2017-05-19:
All the event detector models have been implemented a spec.js file should be created to test adding/modifying/deleting each type of detector.

Puckfist, 2017-05-23:
The POST endpoint /v2/event-detectors calls this.dao.getByXid(xid); which throws a ShouldNeverHappen exception in the EventDetectorDao every time.

Puckfist, 2017-05-23:
Also GET and PUT /v2/event-detectors/{xid} have the same problem

terrypacker, 2017-05-23:
This is being fixed and added to the eventDetector.spec.js file in the mango-node-client

terrypacker, 2017-05-23:
Appears fixed via test spec but could use better test cases in the spec to be sure.

terrypacker, 2017-08-03:
Has test spec, will create new issues if we find problems when we write the UI for this.

closed 2017-08-03 by terrypacker