MangoAutomation/ma-core-public #1016: Timer Tasks/High Priority Tasks that reference Common.runtimeManager can fail

terrypacker, 2017-05-24:
Due to the order of the lifecycle the Real Time Timer is the last item to be shutdown in Mango, which means that any scheduled or submitted tasks to the High Priority Pool will be able to access the various parts of Mango that have been terminated.  In this case the RuntimeManager can return a null reference from Common.runtime manager.

Here is a list of all Tasks that reference the Runtime Manager that should be reviewed:

[ ] - DelayedSetTask
[ ] - PidPwm
[ ] - PurgeAllPointValuesAction
[ ] - DataPurge
[ ] - SetPointWorkItem
[ ] - DataPointImporter
[ ] - DataSourceImporter
[ ] - PublisherImporter

terrypacker, 2017-05-24:
Adding this check to all above code:

		if(Common.runtimeManager.getState() != RuntimeManager.RUNNING)
			return;

jazdw, 2017-06-02:
While doing an update, my backup failed resulting in an active task stuck in the queue on shutdown.
https://gist.github.com/jazdw/fd011758f85001a39672da8d2a658c3a

The process exited after the last log entry.

jazdw, 2017-06-08:
I had a failed upgrade that caused mango to fail starting, afterwards there were stuck threads -
https://gist.github.com/jazdw/6642ef98de125e61b662a4a69a375bee
[image]

terrypacker, 2017-07-31:
I've fixed the problem with the extra threads hanging around on a bad startup, there was a problem with the lifecycle terminating various subsystems if exceptions were thrown.

terrypacker, 2017-08-03:
Confirmed that a failed DB Upgrade check will result in a clean shutdown with no remaining threads as per comment by @jazdw

closed 2017-08-03 by terrypacker