MangoAutomation/ma-core-public #1017: New UI JSON Import Strict vs. Sero JSON format

terrypacker, 2017-05-24:
The import of JSON is more strict in the new UI and forces the JSON to be formatted better than the legacy Serotonin JSON library does.  Both the file upload and the text submit areas use the same service method ImportExport.importData.

If we were to change the 'Content-Type' header to be 'application/sero-json' this will invoke the Serotonin JSON converter and allow less strict formatting through.  

This works fine for the file upload but the text submit seems to also convert the text to JSON prior to submitting to the controller so it still fails.  If we were to just submit whatever text is in the box without conversion to JSON then it would also work for the text input.

I would suggest that we allow the option for importing legacy JSON on the page that effectively changes the 'Content-Type' header between 'application/json' and applicaiton/sero-json'.

Puckfist, 2018-04-23:
Is this more than little things like the permitted comma? i.e.

{ "key": "value", }

Puckfist, 2018-09-11:
We fixed it to handle the Infinity and -Infinity issue. Is anything outstanding besides the extra comma?

terrypacker, 2018-09-11:
Not that I am aware of.