MangoAutomation/ma-core-public #1025: Review Use of ConcurrentHashMap to ensure Atomic methods are used

terrypacker, 2017-05-30:
Review the various uses of this class and ensure things like updating the map are done atomically.  

i.e. to insert a new value if the key is null one should use computeIfAbsent instead of the:
getValue, if null then insert new value logic path.

Puckfist, 2017-08-01:
I changed the dataPointListener map in RuntimeManager to use this, but I did not think it correct to change runningDataSources and dataPoints

Puckfist, 2017-08-01:
https://github.com/infiniteautomation/ma-core-public/commit/d71928fc845b386ed29987b40bca70d1cb9f0c20
https://github.com/infiniteautomation/ma-core-public/commit/35f8acc6b078451457627c594e6814eb47b028b6
https://github.com/infiniteautomation/ma-core-public/commit/b7241d855f0548793033ce0cdd7e80bba3f881e6
checking modules......

closed 2017-08-18 by Puckfist