MangoAutomation/ma-core-public #1027: Remove deprecated install() and upgrade() methods from ModuleElementDefinition

terrypacker, 2017-05-31:
Remove them and replace calls to them with appropriate call during lifecycle.

Puckfist, 2017-08-18:
I think the uninstall method was deprecated in ModuleElementDefinition as well, with the intention of passing a boolean into the postRuntimeManagerTerminate method. Any modules that do work in uninstall probably wish to move it to that method and check the boolean. Presently I don't think anything uses this.

Another option would be to remove deprecation from the uninstall method, since everything uses that already and it doesn't have the same flaw seen in the install and upgrade methods.

Puckfist, 2017-10-03:
This is going to mean storing the loaded versions of modules, right? Then it'll be the database determining if a module's install or upgrade code should execute.

jazdw, 2017-10-04:
I don't believe so? The install/upgrade methods don't know anything about versions, they are just called if there is a zip file on startup.

This issue just means we should change the modules to use a specific lifecycle hook and check the install/upgrade booleans.

Puckfist, 2017-10-05:
But how will you determine its an install/upgrade without having the version stored from after running the hook? The situation is that you unzip it and then Java terminates abruptly. What other than the database can we rely on to resolve if we've run the install or upgrade code? That would also enable an upgrade to know what version it's an upgrade from.

jazdw, 2017-10-05:
So there's two different types of upgrade, there's module lifecycle hooks with an upgrade boolean and there's DB upgrade classes. The lifecycle hooks are only called when a zip file is found.

I think the whole system could use a bit of an overhaul to be honest.

terrypacker, 2017-11-09:
I suggest that we leave the uninstall() method as is since it has special handling to not delete the module if something fails in the uninstall method.

Puckfist, 2018-01-02:
Well, I guess the methods are missing, but I don't think this solved the original issue and the purpose of removing the methods. Iirc, the issue had to do with the internal data source's automatically created data source being done in those methods, but that in the event of either a bundled core with the modules unzipped or a crash during the initial startup that data source was not / will not be created.

The logic for upgrade / install is still only based on the existence of the zip file so I don't believe the actual expression of the issue that brought about this changed. That's the point of #996 but I suppose it'll be as easy as 1) a new database table, 2) judging upgrade / install from that install of the installMap in Lifecycle.

terrypacker, 2018-01-04:
Pushing this to after the release.

terrypacker, 2020-03-24:
Closing this as the remaining portion will be accomplished in #1523

closed 2020-03-24 by terrypacker