MangoAutomation/ma-core-public #1029: Upgrades available endpoint does not give any failure information

jazdw, 2017-06-01:
e.g. If I do a upgrade check when the store is down, I get this printed in my log
```
ERROR 2017-06-01T12:56:39,354 (com.serotonin.m2m2.rt.maint.UpgradeCheck.run:64) - mangoautomation.net:8443 failed to respond 
org.apache.http.NoHttpResponseException: mangoautomation.net:8443 failed to respond
```
but the /rest/v1/modules/upgrades-available endpoint just returns HTTP 500 with no further information
```
{"data":"","status":500,"config":{"method":"GET","transformRequest":[null],"transformResponse":[null],"jsonpCallbackParam":"callback","url":"/rest/v1/modules/upgrades-available","headers":{"Accept":"application/json, text/plain, */*","X-Requested-With":"XMLHttpRequest","X-XSRF-TOKEN":"66f9b7ec-b84e-43de-bdb2-d62ed0d7ffa2"},"cached":false,"timeout":30000},"statusText":""}
```

terrypacker, 2017-06-02:
This may be a point of contention but the v1 endpoints *mostly* use the legacy way of reporting errors, in the 'errors' header.  I just tested this endpoint and it does in fact return this header on error.

If we must change this I would suggest we create a v2 endpoint that uses the new exception stuff.

jazdw, 2017-06-05:
As you can see in the response I posted above there was no extra information in the headers either, but yes we should port this to v2 and get some actual error information into the body.

Puckfist, 2017-08-01:
Easy enough to not just go for. I have put in handling for unknown host and socket timeout. I am sure those two will produce an error header, but this doesn't not seem to be making it to the page still.

terrypacker, 2017-12-01:
Jared can you confirm that this is working and if so add to UI?

terrypacker, 2017-12-04:
Messages are making it back to the page

closed 2017-12-04 by terrypacker