MangoAutomation/ma-core-public #1030: Feature Request - Query on data point by type

WillGeller, 2017-06-02:
Would be useful when building a page when you want to display a list of all binary or all numeric points in the system.

Puckfist, 2017-06-27:
This would not be trivial. Points almost always cannot have their data type resolved unless the blob is deserialized completely. So, any query on this would force loading all points out of the db currently, I think

terrypacker, 2017-06-28:
We have a mechanism for achieving this so it definitely can be done accurately but as @Puckfist said it requires filtering the data after it is returned from the database.  See FilteredQueryStream usage in the DataPointRestController's RQL query methods to filter on permissions after the query is performed.

jazdw, 2017-06-28:
@Puckfist @terrypacker So lets move it out of the blob and put it in a column?

terrypacker, 2017-06-28:
It's a little more complex than that since the dataType is part of the PointLocator, but it also shouldn't be necessary since the FilterStream logic/mechanism is designed to allow for keeping the counts and data returned accurate to the query.  I'm sure there are queries that can't be done this way but we potentially don't need those types of queries.

jazdw, 2017-06-28:
Yeah but that is not going to be practical when you have a large number of data points, anything that we want to be querying on should not be in a blob. Doesn't matter that its part of the point locator, its a 1-1 mapping.

Puckfist, 2017-08-03:
This might be the upgrade that does it: 

package com.serotonin.m2m2.db.upgrade;

import java.sql.ResultSet;
import java.sql.SQLException;
import java.sql.Types;
import java.util.HashMap;
import java.util.Map;

import org.apache.commons.logging.Log;
import org.apache.commons.logging.LogFactory;
import org.springframework.dao.DataAccessException;
import org.springframework.jdbc.core.ResultSetExtractor;
import org.springframework.jdbc.core.RowMapper;

import com.serotonin.m2m2.db.DatabaseProxy;
import com.serotonin.m2m2.vo.DataPointVO;
import com.serotonin.util.SerializationHelper;

public class Upgrade17 extends DBUpgrade {
	
	private static final Log LOG = LogFactory.getLog(Upgrade17.class);

	@Override
	protected void upgrade() throws Exception {
		//Add the data type column into the database.
		Map<String, String[]> scripts = new HashMap<>();
        scripts.put(DatabaseProxy.DatabaseType.DERBY.name(), addColumn);
        scripts.put(DatabaseProxy.DatabaseType.MYSQL.name(), addColumn);
        scripts.put(DatabaseProxy.DatabaseType.MSSQL.name(), addColumn);
        scripts.put(DatabaseProxy.DatabaseType.H2.name(), addColumn);
        runScript(scripts);
        
        //not using data type id to deserialize, so we don't need a legacy row mapper here
        this.ejt.query(UPGRADE_17_DATA_POINT_SELECT, new Upgrade17ResultSetExtractor());
        
        scripts.clear();
        scripts.put(DatabaseProxy.DatabaseType.DERBY.name(), alterColumn);
        scripts.put(DatabaseProxy.DatabaseType.MYSQL.name(), modifyColumn);
        scripts.put(DatabaseProxy.DatabaseType.MSSQL.name(), alterColumn);
        scripts.put(DatabaseProxy.DatabaseType.H2.name(), alterColumn);
        runScript(scripts);
	}

	@Override
	protected String getNewSchemaVersion() {
		return "18";
	}

	private static final String[] addColumn = {
			"ALTER TABLE dataPoints ADD COLUMN dataTypeId INT;",
	};
	private static final String[] alterColumn = {
			"ALTER TABLE dataPoints ALTER COLUMN dataTypeId INT NOT NULL;"
	};
	private static final String[] modifyColumn = {
			"ALTER TABLE dataPoints MODIFY COLUMN dataTypeId INT NOT NULL;"
	};
	
    private static final String UPGRADE_17_DATA_POINT_SELECT = //
    "select dp.data, dp.id, dp.xid, dp.dataSourceId, dp.name, dp.deviceName, dp.enabled, dp.pointFolderId, " //
            + "  dp.loggingType, dp.intervalLoggingPeriodType, dp.intervalLoggingPeriod, dp.intervalLoggingType, " //
            + "  dp.tolerance, dp.purgeOverride, dp.purgeType, dp.purgePeriod, dp.defaultCacheSize, " //
            + "  dp.discardExtremeValues, dp.engineeringUnits, dp.readPermission, dp.setPermission, dp.templateId, dp.rollup, "
            + "  ds.name,  ds.xid, ds.dataSourceType " //
            + "from dataPoints dp join dataSources ds on ds.id = dp.dataSourceId ";
	
	class Upgrade17DataPointRowMapper implements RowMapper<DataPointVO> {
        @Override
        public DataPointVO mapRow(ResultSet rs, int rowNum) throws SQLException {
            int i = 0;

            DataPointVO dp = (DataPointVO) SerializationHelper.readObjectInContext(rs.getBinaryStream(++i));
            dp.setId(rs.getInt(++i));
            dp.setXid(rs.getString(++i));
            dp.setDataSourceId(rs.getInt(++i));
            dp.setName(rs.getString(++i));
            dp.setDeviceName(rs.getString(++i));
            dp.setEnabled(charToBool(rs.getString(++i)));
            dp.setPointFolderId(rs.getInt(++i));
            dp.setLoggingType(rs.getInt(++i));
            dp.setIntervalLoggingPeriodType(rs.getInt(++i));
            dp.setIntervalLoggingPeriod(rs.getInt(++i));
            dp.setIntervalLoggingType(rs.getInt(++i));
            dp.setTolerance(rs.getDouble(++i));
            dp.setPurgeOverride(charToBool(rs.getString(++i)));
            dp.setPurgeType(rs.getInt(++i));
            dp.setPurgePeriod(rs.getInt(++i));
            dp.setDefaultCacheSize(rs.getInt(++i));
            dp.setDiscardExtremeValues(charToBool(rs.getString(++i)));
            dp.setEngineeringUnits(rs.getInt(++i));
            dp.setReadPermission(rs.getString(++i));
            dp.setSetPermission(rs.getString(++i));
            //Because we read 0 for null
            dp.setTemplateId(rs.getInt(++i));
            if(rs.wasNull())
            	dp.setTemplateId(null);
            dp.setRollup(rs.getInt(++i));

            // Data source information.
            dp.setDataSourceName(rs.getString(++i));
            dp.setDataSourceXid(rs.getString(++i));
            dp.setDataSourceTypeName(rs.getString(++i));

            dp.ensureUnitsCorrect();

            return dp;
        }
    }
	
	class Upgrade17ResultSetExtractor implements ResultSetExtractor<Void> {

		@Override
		public Void extractData(ResultSet rs) throws SQLException, DataAccessException {
			Upgrade17DataPointRowMapper dprw = new Upgrade17DataPointRowMapper();
			while(rs.next()) {
				DataPointVO dpvo = dprw.mapRow(rs, rs.getRow());
				if(LOG.isDebugEnabled())
					LOG.debug("Updating dpid: " + dpvo.getId() + " setting data type id to: " + dpvo.getPointLocator().getDataTypeId());
				
				ejt.update("update dataPoints set dataTypeId=? where id=?",
						new Object[]{dpvo.getPointLocator().getDataTypeId(), dpvo.getId()}, 
						new int[] {Types.INTEGER, Types.INTEGER});
			}
			return null;
		}
		
	}
}

jazdw, 2017-10-26:
@Puckfist Think you had an error with the row mapper. Please verify c0528ea70921e1529738e73df14e7b89a32e1e92

closed 2017-12-18 by Puckfist