MangoAutomation/ma-core-public #1035: NoSQL Failed Backdates can get discarded

terrypacker, 2017-06-07:
The insert backdates process submits a list into the NoSQL database that can fail to insert some of the data, upon failure I don't think we know how much of the list was inserted.  We should potentially use a queue to insert the backdates so during failures we don't lose data.

terrypacker, 2018-04-04:
This was potentially fixed in 

commit infiniteautomation/ma-modules-proprietary@72bbe99f586bdb2889251630feee3c586572a32c

and 

commit infiniteautomation/ma-modules-proprietary@2f547dd017e497b79760bdb23eab6e5d5072cb33

Puckfist, 2018-04-09:
Also this one: https://github.com/infiniteautomation/ias-tsdb/commit/8b484c49b42504c01632db679518c98015120d71 to keep the count correct

Then the second commit you referenced uses that count to copy back from the list into the queue

closed 2018-04-16 by Puckfist