MangoAutomation/ma-core-public #1039: MangoES Configuration license needs to be updated

Joelhaggar, 2017-06-20:
This module should only show up available for install if the core is licensed with MangoES or Brewers Dashboard.

Puckfist, 2017-06-27:
The store has the ability to restrict things based on module and account tags. I have a bandaid solution to prevent people from accidentally downloading it applied. Issues may exist if we transfer the license to their account and they delete the module one way or another.

Puckfist, 2017-08-03:
Removing ready for testing as the requested feature, having the restriction structured into the license type, isn't done yet. But, i'm not sure it will be. This issue could probably considered whenever the deployment of the full semver store is.

Puckfist, 2017-10-11:
I fit this in. Module tags for accounts now also factors in the core license type, such that it appears the account also has the tag that is the type of the license for that instance.

closed 2018-01-23 by Puckfist