MangoAutomation/ma-core-public #1041: Missing defaults for system settings

jazdw, 2017-06-23:
I noticed today on a clean database that the system event alarm levels on the new System Settings page were all coming as being "None", upon looking into it I found that there are no defaults registered for these system settings so they all return 0.

I will update this list if I find any more

- [ ] System event alarm levels

terrypacker, 2017-12-22:
What about DataPoint, DataSource, Publisher types?

Puckfist, 2018-01-02:
^ I'm not sure I understand that question. Are you asking if there are always defaults provided for event types for those things?

Data Point and Data Source  events are generally handled by the data source. The createEventType method in DataSourceVO supplies the default alarm level - URGENT. EventTypeVO's cannot be created without an alarm level, so the publishers seem to call getAlarmLevel in PublisherVO wish takes a primitive int for a default alarm level and has no fellow method without a default.

Puckfist, 2018-01-02:
Also, do we care about redundant references to the default values? For instance, in the NoSQL modules, almost all calls to get system settings values include the default value in the call, but all the defaults have already been added to the default value map!

Do we have any sense of idealized values to reason from on keeping or removing those defaults in the calls when the default in the call matches the default in the map?

terrypacker, 2018-01-04:
Don't currently care about redundant references.  Sounds like my concerns are addressed then.

closed 2018-01-04 by terrypacker