MangoAutomation/ma-core-public #1042: H2 database restore does nothing

jazdw, 2017-06-29:
Windows
H2 database
Running restore from REST API which calls
`ProcessResult result = DatabaseBackupWorkItem.restore(filename);`

Nothing in my console. REST returns HTTP OK but the database is not restored.

Puckfist, 2017-07-10:
I tested it, seems to work to me. I assume this was through the /v2/actions/trigger/{action} endpoint. One thing I found weird is that DatabaseBackupWorkItem.restore is expecting a filename within the configured backup directory. So, it took an object like this one in the body for me: {"filename":"core-database-H2-Jul-10-2017_132152.zip"}

Puckfist, 2017-07-10:
Kinda interesting one cannot get the type of system action back out of a system action object.

jazdw, 2017-07-10:
Yeah you need to supply the filename to restore and this works, I traced the filename getting from the REST end point through to the DatabaseBackupWorkItem.restore(filename) but it just didn't seem to do anything.

Puckfist, 2017-08-15:
I just tested the modules page again a few times. It's working. I'm closing.

closed 2017-08-15 by Puckfist