MangoAutomation/ma-core-public #1047: NoSQL retry delete and purge

Puckfist, 2017-07-06:
Because data can be chronologically disordered and a callback must be abandoned, it was made possible for a delete or a purge to fail because a point will have this retry error. A delete or a purge should probably always be retried, and only the number from the second count should be totaled.

Puckfist, 2017-08-03:
The current solution could be improved slightly, as the notion of retries is directed more toward retrying the same shard several times whereas in the nosql dao the retries is for the whole query, so if two or more shards in the query period have disordering corruption, the exception will still make it back out to the user.

terrypacker, 2018-04-04:
Likely fixed by: 3d0ab3029dc651ca709d9540e0f1f27586207a44

terrypacker, 2018-07-16:
Closing as fixed until we see another problem from this.

closed 2018-07-16 by terrypacker