MangoAutomation/ma-core-public #1048: Duplicate usernames can be created

jazdw, 2017-07-07:
There is no SQL unique constraint for username in the users table. Also it doesn't seem like the REST API checks this.

We had a customer that accidentally created a duplicate admin user due to LastPass over-enthusiastically filling the username field.

Puckfist, 2017-07-10:
Related to this item: https://github.com/infiniteautomation/ma-core-public/issues/737

It is significant. One can't log in with a user if there're duplicated I believe, something about queryForObject expecting only one result.

Puckfist, 2017-08-29:
MySQL, H2: ALTER TABLE users ADD CONSTAINT username_unique UNIQUE(username);

Should be perfectly safe since if there were a duplicate user it wouldn't be able to log in and things wouldn't have been working anyway.

Puckfist, 2017-10-03:
Could probably have a safety check first...

List<User> users = UserDao.instance.getUsers(); //we know they're sorted by username, so duplicates should be sequential.
if(users.size() != 0) {
    String last = users.get(0).getUsername();
    for(int k = 1; k < users.size(); k+=1) {
        if(last.equals(users.get(k).getUsername()) { /* do stuff, delete user */ }
        last = users.get(k).getUsername();
    }
}

terrypacker, 2017-12-19:
Fail upgrade if there are multiple users with the same username choose one to save and remove the others (lowest ID kept).  Spit out some obvious messages about the lost users.

terrypacker, 2017-12-20:
This is in Upgrade21 and the create tables sql files.

terrypacker, 2017-12-21:
Tested in MySQL and H2.

closed 2017-12-21 by terrypacker