MangoAutomation/ma-core-public #1051: Strange validation creating data points via REST

jazdw, 2017-07-12:
If I don't supply a templateXid I can create a data point using a minimal point template like so
```
{"xid":"DP_14","dataSourceXid":"DS_HAY_TEST","name":"Display name for Point 14","deviceName":"DS Two","pointLocator":{"modelType":"PL.HAYSTACK","dataType":"BINARY","id":"14"}}
```
However if I set the template XID, I get the following validation error messages back
```
"validationMessages":[{"message":"Required value","level":"ERROR","property":"textRenderer"},{"message":"Invalid value","level":"ERROR","property":"chartColour"}]
```

Puckfist, 2017-08-01:
Can you retest, or at least check if your template somehow has a null chart colour? I was not able to reproduce, as my templates all had "" and importing them with chartColour null still saves them as "" so yours must be holdovers or I am missing something.

terrypacker, 2017-12-13:
Is this important anymore since we have a v2 Data Point controller?

jazdw, 2017-12-13:
I think I fixed this while working on the V2 data point REST controller as I changed the way defaults are set etc. I'll add it to my list of things to check.

terrypacker, 2017-12-13:
We should probably both go over this, I just took a look and the code doesn't seem to use the DataPointDefaulter which leads me to believe there may be other differences to the way the DWR would save points.  I think in order to use the Defaulter we may need to make some core changes.  So we should look at this sooner than later.

terrypacker, 2017-12-13:
When done open a new issue around Data Point Defaults.

closed 2018-01-11 by terrypacker