MangoAutomation/ma-core-public #1059: Cleanup of SystemSettingsDao

jazdw, 2017-08-03:
Some methods are static, some are not - make consistent, all methods should be called on an instance of SystemSettingsDao.

Mark get() methods with default parameters as deprecated to ensure that defaults are added to the DEFAULT_VALUES map.

Puckfist, 2018-04-27:
It's not actually ready for testing. While the methods are no longer static there are still modules calling the methods that take default arguments if their default argument doesn't match what they passed into the default values map. Also there are very probably more static methods in the SystemSettingsDao that could / should be converted to referring to 'instance' (perhaps, almost all of the static functions?!?!)

Puckfist, 2018-05-01:
I didn't mark the methods with defaults deprecated and I didn't eliminate all references to the methods with defaults, only the ones with redundant defaults to what should have been in DEFAULT_VALUES

terrypacker, 2019-02-27:
Closing as methods have been deprecated.

closed 2019-02-27 by terrypacker