MangoAutomation/ma-core-public #1061: Meta Data - return without new value

Puckfist, 2017-08-03:
Currently, if you are in a meta point and you call "return;" you will get the casting of a null into your data type as a default value. The only way to not get a value is to throw "an error that prevents a value."; but this forces you to ignore script errors or you get spammed in regular operation. It would be nice if there was a way for meta points (or point links or set point handlers) to assess at runtime not to generate a new value.

Puckfist, 2017-08-15:
I implemented this with a static data value in the context signifying a unchanged result.

Puckfist, 2017-08-15:
I will probably add something like this to the point links too

Puckfist, 2017-08-17:
This is in the meta module currently, but the way it works was refactored to the core so that the set point handler and point links could also do the same thing. The meta module will want to get updated for the 3.3 release to get this functionality from the core (which is also when it will be added to point links)

Puckfist, 2017-10-18:
This should be ready for any final testing.

jazdw, 2017-10-18:
Could also check if the result of the function is undefined - https://stackoverflow.com/questions/29277606/differ-null-and-undefined-values-in-nashorn

closed 2017-12-15 by Puckfist