MangoAutomation/ma-core-public #1063: Invalid CSRF Token When Switching from HTTPS to HTTP

terrypacker, 2017-08-04:
I experienced a scenario where I couldn't login to Mango because of an invalid CSRF Token.  The error response on the page was:

`An error occurred while logging in - Could not verify the provided CSRF token because your session was not found.`

I accidentally had https on the URL but Mango was not running on https at the time so I got an invalid protocol response so I then changed the url to be http and went to the login page.  At this point I could not login due to the above response even when I refreshed the page multiple times.

I believe this is because at some point in the past I had run a Mango on this port on https and the cookie being sent by the new non-https Mango could not overwrite the existing https cookie.

jazdw, 2017-08-14:
> I believe this is because at some point in the past I had run a Mango on this port on https and the cookie being sent by the new non-https Mango could not overwrite the existing https cookie.

This is correct. See #963

terrypacker, 2017-12-01:
This can be closed when #963 is closed.

Puckfist, 2018-04-05:
Something to keep in mind about this is that even though it has been fixed, if you launch an older Mango on HTTPS you will still be able to get into this condition

closed 2018-04-12 by Puckfist