MangoAutomation/ma-core-public #1066: Able to delete module dependency

terrypacker, 2017-08-04:
I found that I can delete the mangoUI module even when I have modules that depend on it.  This should be fixed for REST and DWR.

Puckfist, 2017-10-12:
I was not able to add a message that makes it back to the page easily, but I did make it such that marking a module for deletion that can't be deleted from a dependency standpoint logs a warning and doesn't mark it.

Puckfist, 2017-10-16:
It is not a 400 response in that situation, with a relevant message in the "localizedMessage" attribute of the response

terrypacker, 2017-12-22:
The backend code works but the new UI doesn't tell you it won't delete it.  However it also doesn't show that it's marked for deletion.  The Rest controller throws:

```
com.infiniteautomation.mango.rest.v2.exception.ModuleRestV2Exception: Dependencies prevented marking for deletion
```

terrypacker, 2017-12-22:
Closing this and opened an issue in the dashboards repo.

closed 2017-12-22 by terrypacker