MangoAutomation/ma-core-public #1069: HTTP Json Receiver - Not ideal

Puckfist, 2017-08-15:
The HTTP json receiver requires that the json object one is decoding is encoded in a regular style parameter field... so a post body like

parameterKey={"nestedObject":"when context type is txt"}

is the desired input currently. Everyone would find in more intuitive if this data source received application/json and could parse values out of a big body, like,

{"body":"just json, no parameters!"}

jazdw, 2017-08-18:
Yep, use Jackson and JSON pointers as per MQTT data source. We should probably store a static ObjectMapper instance somewhere in Core for these purposes and share it with MQTT etc. ObjectMapper 
 is apparently expensive to create. Don't use the existing one that we use for the REST API as they should be configured independently.

terrypacker, 2017-08-18:
Also I noticed that the WatchListDao has its own Object Mapper and I'm about to create another one in the Advanced Scheduler module.  I'm opening this issue for 3.3.0: #1075

Puckfist, 2017-09-14:
Yeah, I used one in HttpDataSourceServlet

There is now a "Receive type" option on http json receiver data sources

closed 2018-03-12 by Puckfist