MangoAutomation/ma-core-public #1070: HTTP Json Retriever - new data source type

Puckfist, 2017-08-15:
We do not have a way to retrieve an HTTP page, then parse values out of it using JSON accessors currently

jazdw, 2017-08-18:
Doesn't really need to be a new data source type, just another option on the point locator perhaps.

From what I understand the data source specifies the URL which is a bit odd IMO, could the URL not be a part of the point locator? You would of course need to check each data point and get a set of URLs to retrieve so you don't make the same HTTP request 10 times but that should be pretty easy to do.

Puckfist, 2017-09-15:
An upgrade of a previous HTTP Retriever should definitely be done.

I decided not to rewrite the data source, so the json retriever is another data source that behaves in much the same way, i.e. specify a URL and use points to identify content within that resource.

Puckfist, 2018-01-02:
This was released in October

closed 2018-01-02 by Puckfist