MangoAutomation/ma-core-public #1076: DaoRegistry - consistence quest

Puckfist, 2017-08-21:
The class DaoRegistry keeps some final static references to the final static singletons in the daos. These are not consistently used as the proxy for the singleton in the code! The vast majority of the time this won't make any difference in a real system, but as the DaoRegistry suggests it could be useful for testing.

Also, perhaps more importantly, there is not consistency in using the PointValueDao stored in the DaoRegistry vs. getting a new point value dao from the database proxy. There is logic surrounding some of the nosql settings that relies on getting system settings out when the dao is created, because they were intended to be short lived. However, many things reference the DaoRegistry's pointValueDao. This is probably not a major issue at the moment, but it is insidious. Since the settings are concerning purges, and the purge doesn't use the DaoRegistry's pointValueDao, we're probably fine on that front. Perhaps the DR's PVD should be used wherever only selection occurs? Perhaps not at all.

terrypacker, 2017-08-21:
I'm of the opinion we can/should get rid of the Registry all together.  It was designed for testing but the whole Mango test framework was never finished and now we just have inconsistencies from it.

Puckfist, 2017-08-22:
I removed these references from all the modules.

Puckfist, 2017-08-23:
I removed references in the core except for the reference to the point value dao in PointValueCache because as written they all share one object (plenty happily) but if modified to eliminate DaoRegistry there is not another place to reuse point value daos

Puckfist, 2017-08-25:
It was determined we could create a static member of the point value cache and just comment the concerns on it, to have both the efficiency and still remove dao registry

closed 2017-10-19 by terrypacker