MangoAutomation/ma-core-public #1077: Create Common Interfaces

terrypacker, 2017-08-24:
To aid in testing we should use Interfaces for the Common members for the 3.3 release:

IReloadingProperties
IRuntimeManager
IDatabaseProxy
IBackgroundProcessing
IEventManager
IRuntimeManager
ISerialPortManager

terrypacker, 2017-10-19:
Do we want to add the ILifecycle interface into the DatabaseProxy (enabling safe mode support)?

Puckfist, 2018-01-02:
While ultimately it's the same value, I think the things that currently refer to RuntimeManagerImpl.RUNNING should refer to RuntimeManager.RUNNING

Other than that I feel contented closing this as tested through every other avenue.

terrypacker, 2018-01-04:
I don't see that in the RuntimeManagerImpl?  The only place RUNNING is defined is in the ILifecycle.  Were you referencing a different interface?

closed 2018-01-12 by terrypacker