MangoAutomation/ma-core-public #1079: Event Detector Rest - Name and alias bad behavior

Puckfist, 2017-08-30:
"name" is not something you can set for an event detector anywhere in the UI. AbstractVOs have names. Event Detectors have aliases. This is apparent in the setAlias method for AbstractEventDetectorVO:

public void setAlias(String alias) {
		this.name = alias;
	}

but for some reason both name and alias are in the models returned from the API. There is validation that only seems to apply to the API that the alias / name cannot be blank. So, you can create a very basic event detector in the ui, GET it, try to PUT it, and you won't be able to because it will fail this validation.

jazdw, 2017-08-30:
Adding @JsonIgnoreProperties({"name"}) to AbstractEventDetectorModel will probably solve this.

terrypacker, 2017-08-30:
I'd like to get away from alias in the new UI and just use name.  Since we aren't using these endpoints yet we should rework the model to only have a name and then remove the alias from the VO (with an upgrade) or set both name and alias to the same thing.

jazdw, 2017-10-18:
I've removed the alias from the model and deprecated the getAlias() setAlias() methods of the VO 221ced183a0aab878f0625bd16fc611e5f9327b1

terrypacker, 2018-04-06:
AbstractEventHandlerVO has the same problem.

terrypacker, 2018-04-10:
One remaining question is how to handle the JSON emports.  I suggest that we begin writing out 'name' instead of alias and provide the ability to read in 'alias' and 'name' with 'name' taking priority if they are both there.

terrypacker, 2018-04-10:
I have updated AbstractEventHandlerVO to deprecate alias and removed it from the model.

Puckfist, 2018-04-16:
https://github.com/infiniteautomation/ma-core-public/commit/c4411ec18d83bea03b267c26b01a297512c12235

closed 2018-04-16 by Puckfist