MangoAutomation/ma-core-public #1083: Need public methods for enable/disable/restart of data points and sources

jazdw, 2017-09-05:
I just added a REST end point for enabling/disabling/restarting data points - `/v1/data-points/enable-disable/{xid}`

However the way it works is not ideal, I have to call `DataPointDao.instance.getFull(dataPoint.getId())` then `Common.runtimeManager.saveDataPoint(dataPoint)`. There should be public methods in the runtime manager to stop and start a data point. There is a public method to restart one.

Data sources should also have these methods.

Puckfist, 2017-09-07:
Could probably just call DataPointDao.instance.getById(dataPoint.getId()) if you didn't want to join the event detectors and user comments. This seems to me like it was somewhat purposeful, perhaps to avoid having to check if dataPoints.contains( dataPoint.getId() ); while starting data points.

The RuntimeManagerScriptUtility uses XIDs, but it goes through the pathway you describe too (exception not calling getFull, just getByXid

Not sure why data sources should have these methods. Elaborate?

Puckfist, 2017-10-05:
Added data source endpoints, modified the data points version some

jazdw, 2017-10-05:
Made a small fix to this infiniteautomation/ma-modules-public@7666ca89c23fa8d95ab70a361ae9b2c46aa2fa4c

closed 2017-12-21 by Puckfist