MangoAutomation/ma-core-public #1086: Add user comments into the events model

jazdw, 2017-09-12:
Currently the events table does a separate REST request to get the user comments for every row in the table.

We should add an option to the events REST controller to get the user comments along with the events.

terrypacker, 2017-12-21:
Changes reverted, this was duplicating work that was already being done in the row-mapper, basically this was slowing down the queries because it was:

1. duplicating the get comments query
2. trying to get comments for EVERY event, which essentially makes a second query for every event returned.  The logic in the row mapper checks the boolean in the EventInstance before attempting to make a query.

terrypacker, 2017-12-21:
So the reason the UI is making a separate call is a mystery to me, the comments have always been returned with events if they have them.

closed 2017-12-21 by jazdw