MangoAutomation/ma-core-public #1088: Logging Data source - NPE for null event messages?

Puckfist, 2017-09-14:
Got a stack trace like:

ERROR 2017-09-14 12:45:17,419 (org.springframework.web.servlet.mvc.method.annotation.ExceptionHandlerExceptionResolver.doResolveHandlerMethodException:367) - Failed to invoke @ExceptionHan$
java.lang.NullPointerException
        at com.infiniteautomation.log4j.Log4jDataSourceRT.loggingEvent(Log4jDataSourceRT.java:71)
        at com.infiniteautomation.log4j.Log4jDataSourceAppender.append(Log4jDataSourceAppender.java:47)
        at org.apache.log4j.AppenderSkeleton.doAppend(AppenderSkeleton.java:251)
        at org.apache.log4j.helpers.AppenderAttachableImpl.appendLoopOnAppenders(AppenderAttachableImpl.java:66)
        at org.apache.log4j.Category.callAppenders(Category.java:206)
        at org.apache.log4j.Category.forcedLog(Category.java:391)
        at org.apache.log4j.Category.log(Category.java:856)
        at org.apache.commons.logging.impl.Log4JLogger.error(Log4JLogger.java:257)

From some older code. I believe the dsrt's 71 is currently 89 and the NPE is probably the event.getMessage().toString() but I don't know a way to generate a log event with a null message offhand.

Puckfist, 2017-09-14:
This may have been triggered by the concurrent modification exception in #1087

terrypacker, 2018-04-04:
Added null check.

closed 2018-04-04 by terrypacker