MangoAutomation/ma-core-public #1089: Reset your password email option

jazdw, 2017-09-15:
We should add a "forgot your password?" option. It should email a link to the user which they can click to reset their password. This could be implemented easily using a JWT with a short expiry time.

Puckfist, 2017-10-12:
I think we need to give them the option to supply an SMTP server at the same time, probably.

Puckfist, 2017-10-26:
Agreed, bad idea to let them supply the server. We should probably hardcode a server it uses if there isn't one configured?

terrypacker, 2017-10-27:
Maybe we only show the option to reset password if a mail server is configured in the system settings?

terrypacker, 2017-10-27:
Also we probably want to think around the logic of how this works if the server fails to send an email, especially if its an admin.  For example if the only admin tries to reset their password but the smtp configuration is bad we don’t want them to end up unable to login...

jazdw, 2017-10-27:
> We should probably hardcode a server it uses if there isn't one configured?

Nope, that's a bad idea. The user has to configure their own, every SMTP server out there requires authentication.

> Also we probably want to think around the logic of how this works if the server fails to send an email, especially if its an admin. For example if the only admin tries to reset their password but the smtp configuration is bad we don’t want them to end up unable to login...

We are not running a web service, the administrator is ultimately responsible for their own instance of Mango and should have access to the backend.

Puckfist, 2017-11-13:
With access to the backend you can already reset your password

terrypacker, 2017-12-04:
I'm seeing this a needing some core API changes that we should put into 3.3.0 and then we can implement a v2 REST endpoint to allow the reset/send the email.

1.  Write up an HTML and Text FTL template for the email
2.  Create a work Item in the core that will send the email.
3.  Create the backing logic/storage for generating random single-use token that is tied to a specific user account that will have its password reset.

The main question is to use a new database table for the token storage or should we simply us a map in memory that goes away if Mango is reset.  Generally these are stored in a database table.

jazdw, 2017-12-05:
> The main question is to use a new database table for the token storage or should we simply us a map in memory that goes away if Mango is reset.

The nice thing about using JWT for this is that the tokens do not need to be stored anywhere. They are cryptographically signed and have an expiration time. When the user clicks the link containing the token as a query param the server reads the token and verifies its signature and expiration date then resets the password.

terrypacker, 2017-12-22:
This is very nice.  Tested using the reset link in the email and also via entering the token into the box.

closed 2017-12-22 by terrypacker