MangoAutomation/ma-core-public #1091: Advanced scheduler websocket does return "active" correctly

jazdw, 2017-09-18:
To test this out, observe the active status of the schedule when enabling and disabling a schedule in the UI. You can inspect the websocket frames to confirm.

When going from active -> inactive the websocket returns the correct result, when going from inactive -> active it still says it is inactive.

terrypacker, 2017-10-11:
I'm not seeing this anymore on the 'schedules' web socket for messages of type rtDataSaved

jazdw, 2017-10-11:
@terrypacker Just checked on this. I didn't realize there was a "rtDataSaved" event type not just "updated". So I'm watching that now and it seems to be OK for when a schedule becomes active/inactive. However you still get what I described when enabling a schedule which should be currently active. I only get an "update" event with "active: false"

jazdw, 2017-10-11:
Tested, fix works for me.

Commit for listening to rtDataSaved event - 4126fbbf490dcb84fb3b4b4eeb658d7733a79c7d

closed 2017-10-11 by jazdw