MangoAutomation/ma-core-public #1092: Advanced scheduler saves rule sets when saving schedules

jazdw, 2017-09-19:
Saving a schedule via the REST API should not save the rule sets included in the JSON body to the DB, it should only validate that the exceptions reference existing rule set XIDs and update the join table.

terrypacker, 2017-10-11:
@jazdw pretty sure this isn't happening but will need to discuss this with you.

jazdw, 2017-10-11:
Yeah just verified that it doesn't save the rule sets. It does however return whatever rulesets I send via PUT back in the response, e.g. I get a schedule, change one of its rulesets name, do a PUT and I get the same ruleset back with the invalid name.

It seems like even if I send a ruleset with an invalid XID (the ruleset XID doesn't exist) it doesn't give an error, it just says 200 OK and returns the same ruleset with invalid XID back to me.

I would expect this behavior if I do a PUT/POST -
1. Server loops over the exceptions and gets the rule set XIDs
2. Lookup each ruleset from the data base by XID
3. If the ruleset XID doesn't return an error response
4. If the ruleset does exist, replace the ruleset on the schedule with this one
5 Return the updated schedule