MangoAutomation/ma-core-public #1093: Advanced scheduler - deleting rule set leaves schedules referencing non-existing rule sets

jazdw, 2017-09-19:
When deleting a rule set any rows in the join table that reference it should be removed. This means that exceptions for a schedule will be removed when the rule set they reference is removed.

terrypacker, 2017-10-11:
I have played around with this one and I'm not sure this is a backend problem.  There is on delete cascade SQL for the create tables and when deleting a rule set the join entry is deleted.  However if I do not refresh the page the other schedules that use this rule-set in their exceptions display an empty entry in the Select rule set drop down.

closed 2018-11-06 by terrypacker

jazdw, 2018-11-07:
@terrypacker do we have a commit for when this was fixed? Was it a backend issue?

terrypacker, 2018-11-07:
@jazdw I closed this assuming it was complete because it was marked ready for testing in April and no further action had been taken.  Usually when I mark something ready for testing and then it gets released I close the issue and assume that if it shows up again we can re-open it.

But to answer your question, it must be (have been) a UI problem as the delete on cascade would enforce the join table's entry to be deleted.

jazdw, 2018-11-07:
Righto good enough for me