MangoAutomation/ma-core-public #1095: Point hierarchy - Not thread safe, dao could use improvment

Puckfist, 2017-09-20:
The point hierarchy is cached in the data point dao. This cached hierarchy is passed modifiable to all takers. The save method is not synchronized, and id assignments can compete. After saving, the cached hierarchy is a new object to be passed to all takers. The save method takes a PointFolder as its argument, but deletes everything from dataPointHierarchy! If lots of threads are concurrently updating and saving the hierarchy, some updates are virtually guaranteed to be missed, and some points may end in the wrong folders.

Puckfist, 2017-09-20:
It would probably not work right if root doesn't have an ID of 0 when passed into savePointHierarchy as written.

Puckfist, 2017-10-05:
I would think we may want to synchronize the savePointHierarchy method and change its argument to be a PointHierarchy

terrypacker, 2017-11-02:
The PointValueDao.savePointHierarchy method is slow in H2, tested removing the pointFolderId index in the data points table but that made it slower.  I was testing with 1000 data points during a JSON emport with each data point having a 'path' set.

terrypacker, 2017-11-02:
Basically I agree that we should synchronize that call but we need to take into consideration the slow performance of that method.  If we synchronize it we probably should only call it from threads that won't lock a UI requests via DWR or REST.  So create a timer task to perform the save...

terrypacker, 2017-11-02:
After further investigation I think we should rework this to use a read write lock.  I can see various locations where modifications are being made to the cached copy.  Perhaps we should create a PointHierarchy singleton with the lock and allow it to perform the various operations with the locks acquired.

terrypacker, 2017-11-03:
Changes we need to make:

- Make save point hierarchy method synchronized
- go back to clearing cache when saving a data point
- When saving always create new and replace cache
- Create a read only point hierarchy and folder to use in other code (or comment modification methods)
- Move DataPointVO.processHierarchyPath() into the Data Point Dao.
- Remove hierachyPath from DataPointVO, keep in model.  When saving apply to hierarchy and when sending out get from hierarchy
- DataPointDao.getPointHierarchy(useCache) Always returns the cached copy...

terrypacker, 2017-11-09:
Should we still export the 'path' in the Rest Model as read only or leave it out?  On very large systems it could degrade performance as we will need to compute the path when we send out the point.

Puckfist, 2018-01-02:
I think the two remain unchecked boxes are no relevant any longer.

1) That method was removed. The import task handles the paths.
2) Cannot be imported through REST, but can be exported.

To the outstanding question, I'm almost indifferent if it's removed or not, but I would lean toward removing it since it cannot be imported. The DataPointVO omits it if the system setting to export it isn't true: if(SystemSettingsDao.getBooleanValue(SystemSettingsDao.EXPORT_HIERARCHY_PATH)) so perhaps offering a similar ability to omit it from the REST is the middle ground.

Puckfist, 2018-01-02:
The second part of that comment ^ may be more pertinent to #1113. I would be content to close this one, with perhaps one more question put out there:

DataPointDao savePointHierarchy takes a PointFolder as its argument. If it took a PointHierarchy, I would find this more appealing linguistically / intellectually, and one could check if hierarchy == cachedPointHierarchy if one wished to get assertive about the readOnly property being enforced at all.

But, I think all current calls correctly identify their intentions to change the PH or not, and the folder enumeration issue is gone, so I think this issue is resolved.

terrypacker, 2018-01-04:
I agree that it would be nice to change the DataPointDao.savePointHierarchy() to take a full PointHierarchy, but its not necessary and we will hopefully moving to a tagging structure soon.  Calling this good for the release.

closed 2018-01-04 by terrypacker