MangoAutomation/ma-core-public #1096: No web socket events for event handlers

jazdw, 2017-09-20:
The websocket exists but is never notified as the Dao has custom save/delete methods which do not call the superclass method.

terrypacker, 2018-03-01:
The REST controller calls the correct methods but none of the Core code does.  Was this issue only about the REST code or is there concern that things like the JSON Importer do not notify the web socket?

jazdw, 2018-04-05:
The REST controller calls save() but this will result in addEventHandlerMapping() not being called for new event handlers. This is a problem.

We should override the save() method to perform the custom logic (create tx and call addEventHandlerMapping()) and remove the custom methods in EventHandlerDao.

terrypacker, 2018-04-10:
The EventDetectorDao will need the same work done to it, ensure we use transaction for updating multiple tables.

terrypacker, 2018-04-11:
This is ready for testing, I've set things up so you MUST call dao.saveFull() to get the ability to add event types when saving either of these.  I've checked the references to both DAOs and made the appropriate changes.  

Testing should involve:

- Create Event Detector from the legacy UI
- Create Event Handler from the legacy UI (See #1243)
- Delete both and confirm no mappings remain
- Export/Import event handlers from legacy UI

- Export/Import data points and assign detectors with handlers (list of xids). I was a little confused on how this should work as currently if you supply detectors.handlers no mappings were previously created.  I assume we want that so I added some code on line 1826 of DataPointVO to make this happen.

terrypacker, 2018-04-11:
Just to confirm the JSON options for this are:

- Export Data Points, 'eventDetectors' is a JSON key with no 'handlers' key output inside them
- Export Event Handlers with 'eventTypes'  as the link to data point (or any VO) XID and Detector XID
- Export Event Detectors can be exported but no link to handlers is made (Not a visible option in UI)

- Import Data Points with 'eventDetectors' that may include 'handlers' and XID list of handlers
- Import Event Handlers with 'eventTypes' that link to data point (or any VO) XID and Detector XID 
- Import Event Detectors containing the key 'handlers' which are the XIDs of existing handlers.

This should potentially be put into a help file/page somewhere.

**NOTE:** I believe that currently the import/export of only EventDetectors would fail since the sourceId of the AbstractEventHandler does not get exported.

**NOTE:** The Event Detector Importer will need some work when we define new types of Detectors as it will fail if a dataPointXID key is not found.

terrypacker, 2018-04-11:
Wrote tests in mango-node-client to prove it's working via the REST api.  Ad hoc testing the other DAO changes via JSON emport and legacy UI.

closed 2018-04-11 by terrypacker