MangoAutomation/ma-core-public #1107: Nashorn -scripting mode

Puckfist, 2017-10-04:
If one passes the -scripting argument to the NashornScriptEngineFactory they get the various functions described here: https://docs.oracle.com/javase/8/docs/technotes/guides/scripting/nashorn/shell.html

Which seems like it would be most useful for a dynamic interaction with the shell

I think we'd have to do this by changing the ScriptUtils.newEngine to call the NashornScriptEngineFactory directy if we wished to do this, or offer it as an option, similar to (but probably not exactly) this (call getScriptEngine I think, not getEngine): https://stackoverflow.com/questions/26873201/enable-scripting-mode-for-nashorn-in-java

jazdw, 2017-10-05:
Are we using Nashorn for shell scripting anywhere? I don't see the usefulness of these options for a meta data point script for example.

Puckfist, 2017-10-05:
No, we're not. The thought was more along the lines of someone could write a really powerful scripting data source.

jazdw, 2017-10-05:
If we do want to do this it seems we could re-implement ScriptUtils.newEngine() as follows. It also enables ES6 features (only let and const in Java 8).

```
    public static ScriptEngine newEngine() {
        ScriptEngineManager manager = new ScriptEngineManager();
        List<ScriptEngineFactory> factories = manager.getEngineFactories();
        for (ScriptEngineFactory factory : factories) {
            if (factory instanceof jdk.nashorn.api.scripting.NashornScriptEngineFactory) {
                return ((jdk.nashorn.api.scripting.NashornScriptEngineFactory) factory).getScriptEngine("-scripting", "--language=es6");
            }
        }
        return manager.getEngineByName("js");
    }
```
Alternatively use command line options `-Dnashorn.option.scripting=false` and `-Dnashorn.option.language=es6`

Puckfist, 2017-10-05:
From a permissions standpoint we may want these options to be supplied by the script implementer, so the scripting data source, meta point, point link, etc such that one script can be making wild shell calls while another script, editable by a lower level user, will only have gaping permissions holes if they write pure Java. There may also be more subtle engine options this makes sense for.