MangoAutomation/ma-core-public #1112: IllegalAccessException trying to run Mango using Java 9

jazdw, 2017-10-05:
Java 9 has been released, we should update Mango to work with it. This is due to using a jdk.internal class. It can be fixed by either
- [Casting to a Runnable rather than a Cleaner and calling run() on that instead](http://mail.openjdk.java.net/pipermail/jigsaw-dev/2016-August/009031.html)
- Using --add-exports on the command line, I suspect we will need this for other nasty old code using sun packages etc

```
ERROR 2017-10-05T17:34:32,406 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:368) - Failed to start data point DP_279767
java.lang.RuntimeException: java.lang.IllegalAccessException: class com.infiniteautomation.tsdb.impl.IasTsdbUtils cannot access class jdk.internal.ref.Cleaner (in module java.base) because module java.base does not export jdk.internal.ref to unnamed module @59ae1544
        at com.infiniteautomation.tsdb.impl.IasTsdbUtils.clean(IasTsdbUtils.java:563) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.ChecksumMappedByteBufferInputStream.close(ChecksumMappedByteBufferInputStream.java:210) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.IasTsdbUtils.closeQuietly(IasTsdbUtils.java:138) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.ReversibleShard.queryReverse(ReversibleShard.java:79) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.CompressibleShard.queryReverse(CompressibleShard.java:387) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.Series.query(Series.java:473) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.tsdb.impl.IasTsdbImpl.query(IasTsdbImpl.java:458) ~[ias-tsdb-1.3.4.jar:?]
        at com.infiniteautomation.nosql.InternalGenericDao.getLatestItem(InternalGenericDao.java:124) ~[mangoNoSqlDatabase-3.2.2.jar:?]
        at com.infiniteautomation.nosql.MangoNoSqlPointValueDao.getLatestPointValue(MangoNoSqlPointValueDao.java:166) ~[mangoNoSqlDatabase-3.2.2.jar:?]
        at com.serotonin.m2m2.rt.dataImage.PointValueCache.refreshCache(PointValueCache.java:185) ~[mango-3.2.1.jar:?]
        at com.serotonin.m2m2.rt.dataImage.PointValueCache.<init>(PointValueCache.java:45) ~[mango-3.2.1.jar:?]
        at com.serotonin.m2m2.rt.dataImage.DataPointRT.<init>(DataPointRT.java:84) ~[mango-3.2.1.jar:?]
        at com.serotonin.m2m2.rt.RuntimeManager.startDataPointStartup(RuntimeManager.java:482) ~[mango-3.2.1.jar:?]
        at com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup(RuntimeManager.java:366) [mango-3.2.1.jar:?]
        at com.serotonin.m2m2.rt.DataSourceGroupInitializer$DataSourceSubGroupInitializer.run(DataSourceGroupInitializer.java:152) [mango-3.2.1.jar:?]
        at com.serotonin.timer.Task.runTask(Task.java:179) [mango-3.2.1.jar:?]
        at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23) [mango-3.2.1.jar:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) [?:?]
        at java.lang.Thread.run(Thread.java:844) [?:?]
```

jazdw, 2017-10-05:
Command line options to workaround this particular exception -
`--add-exports java.base/jdk.internal.ref=ALL-UNNAMED`

jazdw, 2017-10-05:
Also related:
```
INFO  2017-10-05T17:47:33,358 (com.serotonin.m2m2.Lifecycle.initialize:192) - Initializing Spring Object Mapper
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.eclipse.jetty.util.BufferUtil (file:/C:/Users/Jared/test-mango/lib/jetty-util-9.3.11.v20160721.jar) to field java.nio.MappedByteBuffer.fd
WARNING: Please consider reporting this to the maintainers of org.eclipse.jetty.util.BufferUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

jazdw, 2017-10-05:
There are also issues with JSP pages as discussed at spring-projects/spring-boot#10456
```
org.apache.jasper.JasperException: /WEB-INF/jsp/dataSourceProperties.jsp (line: 6, column: 0) The absolute uri: http://java.sun.com/jsp/jstl/core cannot be resolved in either web.xml or the jar files deployed with this application
        at org.apache.jasper.compiler.DefaultErrorHandler.jspError(DefaultErrorHandler.java:41) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.ErrorDispatcher.dispatch(ErrorDispatcher.java:275) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.ErrorDispatcher.jspError(ErrorDispatcher.java:91) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Parser.processIncludeDirective(Parser.java:335) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Parser.parseIncludeDirective(Parser.java:368) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Parser.parseDirective(Parser.java:471) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Parser.parseElements(Parser.java:1435) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Parser.parse(Parser.java:139) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.ParserController.doParse(ParserController.java:227) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.ParserController.parse(ParserController.java:100) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Compiler.generateJava(Compiler.java:199) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Compiler.compile(Compiler.java:356) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Compiler.compile(Compiler.java:336) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.compiler.Compiler.compile(Compiler.java:323) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.JspCompilationContext.compile(JspCompilationContext.java:585) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.servlet.JspServletWrapper.service(JspServletWrapper.java:363) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.servlet.JspServlet.serviceJspFile(JspServlet.java:396) ~[apache-jsp-8.0.33.jar:2.3]
        at org.apache.jasper.servlet.JspServlet.service(JspServlet.java:340) ~[apache-jsp-8.0.33.jar:2.3]
        at org.eclipse.jetty.jsp.JettyJspServlet.service(JettyJspServlet.java:108) ~[apache-jsp-9.3.11.v20160721.jar:9.3.11.v20160721]
...snip
```

jazdw, 2017-10-06:
Also launching Mango using a Java 9 runtime from Eclipse is broken (on Windows at least)
https://bugs.eclipse.org/bugs/show_bug.cgi?id=522333

Puckfist, 2017-10-06:
For dodging the cleaner issue you can use a different database stream type. Only MappedByteBuffer requires the cleaner

Puckfist, 2018-03-19:
Looks like this might be the solution: https://stackoverflow.com/questions/25238110/how-to-properly-close-mappedbytebuffer

jazdw, 2018-04-12:
@terrypacker any reason were not going to fix this issue for Mango 3.4? It really should be fixed ASAP given that Java 10 has now been released.

terrypacker, 2018-04-12:
I'm happy to put it on the list for 3.4

terrypacker, 2018-04-17:
After fixing the Database Cleaner problem and using the latest Eclipse I could launch Mango in Eclipse.   However we still have some outstanding problems:

-  [jetty-alpn-agent] Could not find a matching alpn-boot JAR for Java version: 9.0.4
-  WARNING: Illegal reflective access by org.springframework.cglib.core.ReflectUtils$1 

Looks like our version of Spring needs to be updated to 5+ and the Jetty ALPN agent only works for Java 7 and 8.

Puckfist, 2018-04-17:
I think we're going to see more of those illegal reflective access messages mayhaps. I saw this one building the UI module:
WARNING: Illegal reflective access by com.thoughtworks.xstream.converters.collections.TreeMapConverter (file:/C:/Users/Phillip/.m2/repository/com/thoughtworks/xstream/xstream/1.4.4/xstream-1.4.4.jar) to field java.util.TreeMap.comparator

Perhaps we should put Java 9 compatibility things in their own place?

Puckfist, 2018-04-17:
The cleaner works!

closed 2018-11-06 by terrypacker