MangoAutomation/ma-core-public #1113: Point Hierarchy - Import points with paths instead of forcing hierarchy construction

Puckfist, 2017-10-10:
It has been requested that points be able to import with a "path" property that can specify "path/to/parent/folder" which will then place the point there and create any necessary folders.

Should we call it something else since the data point isn't a part of that path? I was not able to add this to the JSON File import or the Mango API yet as those would prefer to have the core changes available, meaning 3.3 feature. This means that the data point model doesn't do this yet, only the serotonin JSON importer. It's also not in the UI module yet, so that'll be a 3.3 objective too. By default it's disabled in exports and / is the path separator.

Puckfist, 2017-10-11:
I just put some code into the main-parial-hierarchy-updates branch of the core that enables the "path" attribute to behave in API data point vo json exchanges. It could also be added to the CSV. In the API module, all one has to do is add this to the data point model:

@JsonGetter("path")
	public String getPath() {
	    return PointHierarchy.getFlatPath(this.data.getId(), DataPointDao.instance.getPointHierarchy(true).getRoot());
	}
	
	@JsonSetter("path")
	public void setPath(String path) {
	    this.data.setHierarchyPath(path);
	}

terrypacker, 2017-10-23:
main-partial-heirarchy-updates has been merged we will now need to fix up the remaining items from the issue description.

terrypacker, 2017-11-03:
I still need to add the CSV column getter annotations before we can test completely.

terrypacker, 2017-11-07:
We have decided not to allow this from the REST api as it would require locking the point hierarchy during each save of a data point.  The hierarchy should still be imported/modified by the path JSON member in an import.

Other work will be done in #1095

terrypacker, 2017-12-12:
The CSV has been pushed for after the release, see #1172

terrypacker, 2018-01-05:
Test with various hierarchy structures, some invalid and some with multi-nested configs.

closed 2018-01-18 by terrypacker