MangoAutomation/ma-core-public #1123: Multiple DataPointRTs can be started for the same data point.

terrypacker, 2017-10-19:
Due to the lack of locking the RuntimeManager.saveDataPoint() method it is possible to start 2 runtimes for the same data point if the requests are issued close enough together.

terrypacker, 2017-10-19:
Investigate the possibility of locking on a data point's ID when entering the method or terminate/replace the existing runtime while starting a data point.

Puckfist, 2017-11-01:
Here's an option, just replace the dataPoints.put(dataPoint.getId(), dataPoint); with...
```
                    dataPoints.compute(dataPoint.getId(), (k, rt) -> {
                    if(rt != null) {
                        try{
                            getRunningDataSource(rt.getDataSourceId()).removeDataPoint(rt);
                        }catch(Exception e){
                            LOG.error("Failed to stop point RT with ID: " + vo.getId()
                                    + " stopping point."
                                    , e);
                        }
                        DataPointListener l = getDataPointListeners(vo.getId());
                        if (l != null)
                            l.pointTerminated();
                        rt.terminate();
                    }
                    return dataPoint;
                });
```

Since you can't call stopDataPoint (don't make other changes to hashmap whilst computing) but you also don't need to redo the validation in saveDataPoint

terrypacker, 2017-11-28:
One way to test this is by importing data points on a scripting data source that also generates the same points.

closed 2017-12-18 by Puckfist