MangoAutomation/ma-core-public #1129: Wide Query Point Value Time Support

terrypacker, 2017-10-24:
Add a method to the PointValueDao that performs a wide query.  This is a query with the value before the time range and the value after the time range returned with the values within the time range.

terrypacker, 2017-10-24:
Modify PointValueFacade.getPointValues(long, boolean, boolean) to leverage this in 3.3.

terrypacker, 2017-11-03:
We need to leverage this in the rollups for the REST api and Excel reports etc.

jazdw, 2017-11-07:
What is the current status of the PointValueFacade? Is this still being used?

terrypacker, 2017-11-07:
Yes its used in the legacy UI and also in the 'latest-*' REST endpoints.  It is also in the 'first-last' point values endpoint.

terrypacker, 2017-12-12:
Closing this as we are now using 'bookend' queries which are different than a wide query.

closed 2017-12-12 by terrypacker