MangoAutomation/ma-core-public #1133: REST JSON Import times out

jazdw, 2017-10-27:
A HTTP GET request to `/rest/v1/json-emport/import/IMPORT_xxx` will time out when importing point hierarchy into a system with a large number of points. I had to add code to keep retrying the REST end point until it didn't timeout.

This tells me something is locking on the back-end while it is doing it's import process, this should not be the case.

This is the import thread that was chugging away in the background.

[image]

terrypacker, 2017-10-27:
From review of the code it doesn't look like there is any locking, it is just performing a large transaction on the database.  We will need to review the code within the transaction.  It is currently doing this:

1.  DELETE FROM dataPointHierarchy
2. UPDATE dataPoints SET pointFolderId=0
3. For **every data point** INSERT INTO dataPointHierarchy (id, parentId, name) VALUES (?,?,?), ... (?,?,?)
4. For **every folder** UPDATE dataPoints SET pointFolderId=? WHERE id in (?,...,?) 

Note that 4 operates recursively on all the folders.  I wonder if you have a lot of folders too or just a lot of points?

jazdw, 2017-10-27:
There is obviously a lock that the REST controller is trying to get in order to retrieve the status. I don't really care what is going on in the background, my HTTP request should always complete in reasonable time. That is the whole point of scheduling a background task and then retrieving its status via a temporary resource.

terrypacker, 2017-10-27:
Ah sorry, I didn't realize you were getting the status.  Yes there is a lock there when it gets a copy of the response messages. from the ImportBackgroundTask  I'll fix that up to use a better data structure without a lock.

terrypacker, 2017-10-27:
The reason I added a lock was due to problems during testing.  According to line 42 in ImportContext the list of messages should be concurrent via CopyOnWriteArrayList but from experience this didn't work correctly.

To fix this problem we may need to modify ImportContext to allow access to the result messages from multiple threads.  Some ideas are:

1.  review the ImportContext line 42 and test without the lock in the controller.
2.  create a listener pattern to listen to any messages generated.
3.  make ProcessResult.messages a concurrent collection

terrypacker, 2017-10-31:
I've removed the lock but I think there may still be improvements we can make.  I've noticed that every time the UI polls the status endpoint it gets back all the messages about what was created.  When importing 100k points each call gradually gets slower due to the increased payload.

Perhaps a Websocket would be good so that you only have to poll the status endpoint

closed 2017-12-22 by terrypacker