MangoAutomation/ma-core-public #1134: JsonEmportRestContoller vs. ImportTask duplicate code.

terrypacker, 2017-10-27:
- Combine into 1 instance
There is a duplicate of the logic in the ImportTask in the REST api because it was added prior to a Minor Core release.  This is problematic because a new PointHeirarchyImporter was created and added to the ImportTask but not the Rest controller.  This highlights the issue with duplicating code.

The solution is to put the importer definitions somewhere common or share the task somehow.

Puckfist, 2017-10-27:
There is also FileImportTask in the JSON File Importer module. Given that you can upload a configuration through the API we could consider removing the JSON File Importer. But, it is the only thing currently polling for configurations to import.

Puckfist, 2017-10-27:
I believe the changes to the core import task were not breaking to the module variants

terrypacker, 2017-10-27:
I'll try to centralize the Importers into common code so that it can be used everywhere.  That way in the future we won't have to worry about missing out on features added to the core.

terrypacker, 2017-11-01:
We need to check the JsonEmportScriptUtility.ScriptImportTask to be sure it is still functioning properly after removing the runImpl() method since it appeared to be a duplicate of the superclass ImportTask

closed 2018-01-03 by Puckfist