MangoAutomation/ma-core-public #1136: Realtime point status not available via websocket

jazdw, 2017-10-30:
We need to get the realtime point status aka UNRELIABLE/DISABLED or not via websockets. This should probably just be made available via the point value websocket. The point value websocket currently tells me if a point is disabled I believe so it shouldn't be too difficult to add a new event type for unreliable.

Puckfist, 2017-11-07:
For that matter, attributes as a whole aren't given to DataPointListener implementers. Just to briefly consider the larger issue, should a listener be aware of attributes?

Just rambling, but that would be a different kind of notify work item created in the SetAttribute method? That means every update to every point that uses the attribute would generate work items? Not ideal. Could every event notify work item that was an update imply it was not unreliable? Would we republish UNRELIABLE event every poll of a point that would be deemed unreliable?

Maybe instead of setting the attribute directly, we should pass the attribute map to set to the dprt so that it can in turn hand it to the EventNotifyWorkItem it would create with the update? Then launch another event notify work item from the set attribute method?

terrypacker, 2017-11-14:
Currently in some data sources UNRELIABLE is set or cleared every poll for every point.  So if we do want to multicast events out we should add some logic to only fire events on change of attribute state.

I also agree with @Puckfist that you can assume that every time an UPDATE event is fired for a point the point value is reliable.  

However since you can poll the UNRELIABLE attribute via /v1/realtime endpoint is this really necessary?

terrypacker, 2017-12-01:
Any time any of the attributes change send an ATTRIBUTE_CHANGED Websocket event.  Create new interface for this type of listener.

terrypacker, 2017-12-12:
Send all attributes on every change.

terrypacker, 2017-12-13:
This is ready for testing, it works as advertised but we should probably consider/test what happens when the system is loaded with these types of events/messages.

terrypacker, 2017-12-22:
Closed as tested with one listener and with the multicast listener group.

closed 2017-12-22 by terrypacker