MangoAutomation/ma-core-public #1142: Event Detector JSON Import/Export

terrypacker, 2017-11-07:
Current event detectors can be imported if you supply the proper JSON, however this is not obvious as you cannot export them independent to Data Points.

Suggest removing the JSON export from the data points, creating a new block for them, and allow importing of the legacy way OR the new eventDetectors block.

Puckfist, 2017-11-08:
The reason was not wanting to restart a data point with 10 event detectors 10 times. And being legacy compatible.

terrypacker, 2017-11-08:
I envision this to work the same as the new 'path' member of the data point JSON for the point hierarchy.

terrypacker, 2017-11-10:
See issue #1144, Event detectors are exported with the full configuration in 3.3.0

terrypacker, 2017-12-13:
I still think we need to add logic to not restart a data point every time a new event detector is added.  We should leverage the logic from DataPointImporter where it only updates the point hierarchy once.

terrypacker, 2017-12-13:
We also still need to add the Export element for the ConfigurationExportData class.  See line 136 and 137 comments.

Puckfist, 2017-12-15:
In order to only save every data point once you would have to hold the whole import of points and detectors in memory since you wouldn't know if a point had a detector importer coming and similarly the order of the importers work is not guaranteed. To avoid this, I am only tracking data points mentioned by event detectors in the "eventDetectors" list, and then only saving them one addition time to the save in the data point importer (if they are being imported there. In which case, the right way to do the import IMO is to put the detectors in the "eventDetectors" object on the data point.

terrypacker, 2017-12-15:
Another slightly wilder idea could be to import everything into the database and then refresh the Runtime Manager at the end of the entire import.  This would mean that for some time the database and the runtime would be out of sync.  This probably isn't the right strategy but something along these lines might be worth looking into.

terrypacker, 2017-12-22:
So from what I can tell there is the ability to import them but still no ability to export them other than within a data point.  So we need to decide if we are adding the export element to the UI.

Puckfist, 2017-12-22:
I think we decided that we are not. I had already removed it. We'll have to revisit this when/if there are other types of event detectors