MangoAutomation/ma-core-public #1145: Optionally output PointValueTime timestamps as custom formatted strings or epoch millis

terrypacker, 2017-11-10:
Test the various endpoints used by the Watchlist download area.  The new parameter useTimestamp defaults to true and returns the long millis since epoch, if set to false it will use:

CSV - format from env properties rest.csvDateOutputFormat=yyyy-MM-dd HH:mm:ss.SSS

JSON - format from env properties rest.customDateOutputFormat=yyyy-MM-dd'T'HH:mm:ss.SSSXXX

Puckfist, 2018-01-01:
It doesn't seem like the Simplify rollup option in the watchlist download data area did anything yet.

terrypacker, 2018-01-04:
That is expected, the watch lists/data point details still use the v1 endpoint and do Simplify on the page.  The v2 endpoints will be leveraged after the release in a new UI module

terrypacker, 2018-01-04:
Closing as tested but not actually used.

closed 2018-01-04 by terrypacker