MangoAutomation/ma-core-public #1147: New Core Temporal Adjusters Need Testing

terrypacker, 2017-11-15:
There are new temporal adjusters for use with the quantize3 package in the core that need testing.  We should write JUnit tests like the DateUtils tests for these for the 3.3.0 release.

The adjusters are in the com.infiniteautomation.mango.util.datetime package.

terrypacker, 2017-12-21:
There is a start for a test in com.infiniteautomation.mango.util.datetime.TruncateTimePeriodAdjusterTest

Puckfist, 2017-12-21:
I got Jared to agree with me about leap day, but we can continue that discussion. Also note that weeks will go back to the adjusted start of the week, meaning the first day of the week of the year, as opposed to Monday always in the original code. I think this makes sense because the weeks are adjusted, too.

closed 2017-12-22 by Puckfist